thenickdude/wacom-driver-fix #86: Graphire 2 not recognized at all on MacBook Pro 2021?

jboolean, 2022-03-17:
This isn't a problem with your driver, but since you're so deep into it maybe you have an insight.

I have an Graphire 2. On older computers, the tablet is recognized as a basic mouse until the drivers are installed to enable pen features. But on my new MacBook Pro 2021 I can't move or click at all, and the driver says the tablet was not found. Did new Macs drop support for USB 1 or something (even with a dongle)?

thenickdude, 2022-03-17:
No, it should work. Make sure you don't have the newest Wacom driver installed, because it'll ask macOS to assign the tablet to it, which turns off the mouse emulation, but then it'll refuse to do anything else with it.

jboolean, 2022-03-17:
It magically started working. Sorry to bother! 

This is AMAZING. The official driver didn't even seem compatible with my 10 year old MacBook but your driver works on an M1 flawlessly.

closed 2022-03-17 by jboolean