thenickdude/wacom-driver-fix #88: Wacom Cintiq 21UX (dtz-2100) on new MacBook Pro with M1 Max

vladsrynty, 2022-04-12:
First, thank you for the driver fix. I was able to install it on Monterey 12.3.1 running on MaxBook Pro 16'' with M1 Max processor.

But I am still not able to get the picture on the screen. Maybe somebody of you can give me some ideas what to try next.

I am using USB type C to DVI-D comverter from Lindy (AG9310 chipset). Blue LED turns on showing that the signal is there and from mac settings the screen is detected. I can drag the windows to the (invisible) screen. Trying the same converter with an old Dell monitor (E213Hc) works fine.

Apple support tells me that it is Wacom screen's firmware problem. 

Can it be the driver issue? What could be possible wrong with the signal? Should I try some other converter? Why? What else could I try?

thenickdude, 2022-04-12:
This can't really be a driver issue because the Wacom drivers aren't needed for the screen functionality to work (basically because the tablet is a standard monitor, and the drivers couldn't influence its connection to the GPU even if they wanted to).

Does the tablet flash up its "no signal detected" message at least? Make sure power is connected to the tablet.

vladsrynty, 2022-04-12:
Strangely enough I got it working by changing the converter to Lindy USB Type C To HDMI/DVI/VGA (chipsets VL100, IT6562FN, PS8222B) and connected to DVI output. Why remains a mystery.

closed 2022-04-12 by vladsrynty