thenickdude/wacom-driver-fix #89: pressure/eraser/touch not working Mac OS Monterey

littlealliegator, 2022-04-15:
Hi,

First off, thank you for offering this to people! I have a Intuos 3 PTZ-930 trying to run on M1 Mac OS Monterey v12.2.1. I installed the driver per the instructions. First time didn't work, so as instructed I went to "Wacom Tablet Utility." Nothing much showed up there but I hit "remove" under "tablet software:" and then reinstalled the driver. This time I was prompted to go to permissions as it said. After restarting my computer and trying the touchpad again then I was prompted to go to "Accessibility" as it said, so now I have "WacomTabletSpringboard" checked in Accessibility and Input Monitoring.  In System preferences, the wacom tablet program can't sync and says there is a problem with the driver. The only drawing software I have at the moment is Gimp. It recognizes the pen and the clicker on the side of the pen but nothing else is working. First time I opened it, it would show the eraser icon if the eraser was hovered a bit above the tablet but then turn into a pen when it got too close. After closing program/replugging tablet now it is not showing the eraser icon and just stays the pen.

littlealliegator, 2022-04-15:
Ok I got it to synchronize by going to "Security & Privacy"-> "General" and then opening some dialogue at the bottom that said something about items needing permission before running. In the list were a bunch of HP items and one Wacom one, which had all popped us as errors when I did the first driver install. I tried to check all of them but it didn't do anything. So next I just checked the wacom one and now it synchs. System Preferences->Wacom Tablet shows everything and show pressure working, but still doesn't work in Gimp. Tired adding Gimp to "Accessibility" and as "Application" in "Wacom Tablet" but no dice. Maybe its just that software, but not sure what else to test it on. :/

thenickdude, 2022-04-15:
To what extent does it not work in GIMP, are you unable to draw at all or are you just not getting pressure support?

Note that pressure support needs to be explicitly turned on inside GIMP:

https://support.huion.com/en/support/solutions/articles/44002277605-how-to-setup-gimp-pen-pressure-on-macos

littlealliegator, 2022-04-16:
It was drawing just no pressure or eraser. Ahh beautiful I would never have thought you needed to turn pressure on in Gimp! How goofy. Thank you that worked! I also looked it up and got the eraser working by turning the pen over and then selecting the eraser tool with that side. Can't get my touch strip to work (and trying to fiddle with the Wacom settings somehow crashed the synchronization and I had to restart my computer to restore it) but there's keyboard shortcuts for zoom. This is pretty usable for a 15 year old piece of tech, awesome! :)

Thank you so much. I was getting ready to give up and buy a new tablet and begrudgingly pay the PS subscription fee, but now I can do at least some light art. (I miss my old mac; it was a beast but finally died. I have CS5 suite 32-bit so can't install it on the new one and my tablet worked great on it, but oh well you can't hold onto things forever.).

closed 2022-04-16 by littlealliegator