thenickdude/wacom-driver-fix #90: Graphire ET-0405-U eraser function not working on MacOS

iperezpuelles, 2022-04-26:
Hello and thank you for the patched driver for MacOS, works great! but the eraser function is not working.
Got a Mac setup with dual boot on 2 SSD's: MacOS Big Sur + Windows 11
While in Windows (version 6.1.7-3 wacom's original) works fine.
While in MacOS (version 6.1.6-4 thenickdude's patched) works fine except for the eraser function.
Any help on this issue is appreciated, thank you.

[image]
[image]

thenickdude, 2022-04-26:
Hmm, try using Wacom Utility to delete your preferences and see if it fixes it.

iperezpuelles, 2022-04-27:
Thanks for your reply, reset preferences done. The issue persists.
It's the pressure sensitivity setting that doesn't seem to work for the eraser tip on the "Try Here" grey rectangle.
The eraser tip seems to work only like a normal pen click although it is set to "Erase"

[image]
[image]

iperezpuelles, 2022-04-27:
I also tried Photoshop on both Windows and Mac: "all with default settings"
Windows Photoshop: draw with the paintbrush then flip the pen to erase and it erases fine.
MacOS Photoshop: draw with the paintbrush then flip the pen to erase and it just keeps drawing.

thenickdude, 2022-04-27:
Ah yep I can reproduce that problem here too with my Graphire 1, I'll see if I can figure out what's going wrong.