thenickdude/wacom-driver-fix #91: Cintiq 12WX - DTZ-1200 Compatibility

mind-sword, 2022-05-31:
Heyo,
Awesome work you guys are doing. Just wondering if this driver should work with a Cintiq 12WX - DTZ-1200. I've seen someone else ref this driver on reddit for that specific model but not seeing it listed under the install notes. Cheers!

thenickdude, 2022-06-01:
As far as I know Wacom's official 6.3.40-2 driver still works on the latest macOS versions for this tablet (despite not *officially* supporting anything newer than Catalina).

My patched 6.3.15-3 "Intuos 3" driver is compatible with the 12WX.

mind-sword, 2022-06-01:
Appreciate the reply man. Sweet I'll give it a go again. Might have just needed a few extra restarts and install/reinstall of the drivers to finally click. Thanks!

rocioep79, 2022-10-25:
Hi, I have an iMac late 2015...  I'm still keeping catalina iOS to continue using my Wacom Cintiq 12WX :/
I wanted to ask you, did the driver finally work with iMacOS Monterey? Please let us know. I will really appreciate it!
Thank you!

Musi134, 2026-09-19:
oi  eu queria (wacom dtz 1200w driver) vcs poderia me passa por favor