thenickdude/wacom-driver-fix #94: no PTK-440 drive fit with M1 Mac?

Ehahahaha, 2022-07-27:
Wacom's official driver v6.3.41 doesn't work...

thenickdude, 2022-07-27:
Other people report it working fine, try the same sort of permissions troubleshooting that my guide suggests

Ehahahaha, 2022-07-28:
Thanks for answer~Maybe they used older version system? I use MacOS Monterey system and followed guide, but still doesn't work with PTK 440.....I checked the Wacom official website and v6.3.41 only work for macOS 10-11.

thenickdude, 2022-07-28:
That's just what it officially supports. It works fine on Monterey, but it isn't listed there because the driver was released before Monterey existed:

https://www.reddit.com/r/wacom/comments/qg2xgc/fixed_macos_12_monterey_drivers_now_available_for/htarxiw

Ehahahaha, 2022-08-01:
problem solved！thanks a lot～

thenickdude, 2022-08-01:
What was the secret to getting it working?