thenickdude/wacom-driver-fix #98: Ventura Mapping Bamboo CTH-670

danielrdotcom, 2022-11-05:
Everything still mostly works, just get no window pop up for "mapping" the pen area. Bit of an issue with multiple monitors. I know things changed in ventura, wanted to see if there are plans to update for this. Thanks!

Suru83, 2022-11-19:
hi, is there any update on this?

thankyou

suruchika *|* bangalore
<https://www.linkedin.com/in/suruchika-singh-a5a35418/>
<https://twitter.com/notsuruchika>
<https://www.instagram.com/notsuruchika/>

thenickdude, 2022-11-24:
Can you try out this pre-version for me? It should fix this issue:

https://github.com/thenickdude/wacom-driver-fix/releases/tag/patch-10-bamboo

Suru83, 2022-11-24:
> Can you try out this pre-release version for me? It should fix this issue:
>
>
> https://github.com/thenickdude/wacom-driver-fix/releases/tag/patch-10-bamboo

hi,
My Bamboo CTH-670 now it works for Ventura, both for finger touch and the
pen accessory.
But on my other Mac with Monterey, the finger touch works .. not the pen
accessory, what can be done?

thanks!

suruchika *|* bangalore
<https://www.linkedin.com/in/suruchika-singh-a5a35418/>
<https://twitter.com/notsuruchika>
<https://www.instagram.com/notsuruchika/>

thenickdude, 2022-11-24:
That's usually a problem with permissions, try the instructions for removing and re-adding the Input Monitoring and Accessibility permissions.

Suru83, 2022-11-24:
yes it seems to work now. thanks

suruchika *|* bangalore
<https://www.linkedin.com/in/suruchika-singh-a5a35418/>
<https://twitter.com/notsuruchika>
<https://www.instagram.com/notsuruchika/>

closed 2022-11-24 by thenickdude