thenickdude/wacom-driver-fix #101: CTH-470 Not working on Ventura (M1)

KleydsonSPanzariello, 2022-11-12:
I recently updated to Ventura and now my tablet is not working properly anymore. I can still finger-touch and move around, but other than that it doesn't work. Pen touch is unresponsive. Buttons are completely useless now, and the Wacom tab in System Settings is also not working. I reinstalled the driver twice but it still doesn't work. Any help? 

[image]

jodineuf, 2022-11-21:
I have the same issue with my CTL-470 tablet using Ventura (M2)

thenickdude, 2022-11-22:
My Bamboo tablet is working on Ventura. This might be due to broken tablet preferences, try using Wacom Utility to remove preferences for the Current User, and All Users, then reboot and try using the preferences again.

lKutts, 2022-11-22:
> My Bamboo tablet is working on Ventura.
> 
> This might be due to broken tablet preferences, try using Wacom Utility to remove preferences for the Current User, and All Users, then reboot and try using the preferences again.

I had done that during the reinstallation. Didn't solve the problem. My only solution was to downgrade to Monterey unfortunately.

ltwon5, 2022-12-06:
Same here
I jumped on Wacom's customer support chat. I was told that the CTH-470 supports stops at Monterey and won't go into Ventura because it's been 14 years

thenickdude, 2022-12-06:
They say this because this is their official company position (because they no longer want to incur support costs for their older tablets), not because it is literally true. 

My Bamboo tablet is working fine in Ventura, as is my Graphire 1 from 2008.

ltwon5, 2022-12-06:
> My Bamboo tablet is working fine in Ventura, as is my Graphire 1 from 2008.

Great job mate, how did you get the CTH-470 to work in Ventura, I cannot get past the same error in  iKutts' screenshot
using Driver 5.3.7-6 (macOS 10.9 - 10.12) 27 MB 10/12/2016 on Wacom's site

Keen to learn and reduce e-waste : )

thenickdude, 2022-12-06:
You can't use the driver from Wacom's site, that one doesn't even work on macOS Catalina, you need to use my patched driver.

I did nothing special to make mine work, it just works. If I knew why OP's wasn't working I could fix my patched driver accordingly.

lKutts, 2022-12-30:
Just wanted to update since this issue was opened before the November 24 update. 
Everything is working now with this new update. I was going to say that the pen mapping tab was not opening but it just did right now... So yeah... everything is working! Good job @thenickdude! Thanks a lot for the driver! I sent you a tip there on PayPal.

closed 2022-12-30 by lKutts

brlmr11, 2023-01-11:
Hi,

I have the same issue with my CTL-471 tablet using Ventura.What change made to the driver?