thenickdude/wacom-driver-fix #103: Error with Illustrator CC 2023 27.0.0

ckoshikumo, 2022-12-07:
The newest Illustrator (CC 2023 27.0.0) doesn't seem to like the driver. On startup, it shows the error: "Illustrator found an issue with Wacom driver" and directs me here: https://helpx.adobe.com/illustrator/kb/illustrator-cc-crash-wacom-drivers.html

Is there any hope of keep using my brave little tablet?

Bamboo Create, Macbook Air M1, 2020, Ventura 13.0.1.

Thanks.

thenickdude, 2022-12-07:
I think my modified drivers never worked with Illustrator CC, did they work for you at some point? 

I haven't been able to test it because I dont own a license for Illustrator, but I suspect it will only work with officially signed Wacom drivers, which would rule my modified ones out completely.

thenickdude, 2022-12-08:
I'd like to support it but I already used up my trial of Illustrator years ago

pointy91, 2023-05-28:
Illustrator used to work on Monterey with Intuos 3 Pro, but after Ventura I have not been able to get tablet working longer than few minutes, the control panel waits for ages to connect, and Illustrator  23 will pop up wacom error even when tablet is disconnected ( and driver uninstalled)