thenickdude/wacom-driver-fix #104: Only the rubber side works - Intuos 2

cladjidane, 2022-12-09:
Hello, thank you for your work. 
I have a problem, I only have the eraser side working, do you think it comes from the pencil (intuos 2)?
The lead side of the pencil does not work at all

Thank you for your answers :)

thenickdude, 2022-12-09:
It's difficult to tell if it is a hardware problem or a driver problem.

In the Wacom settings, try holding the Alt key on your keyboard and clicking "About" to open the diagnostics screen (or maybe there is a Diagnostics button inside the About menu). In there you can see if when you use the pen tip, it shows the pressure values changing.

cladjidane, 2022-12-09:
Thanks for the quick response!

I understand that it is indeed difficult to judge if it is a hardware problem or not :) But I think about it because here is the diagnostic info on the rubber side: 

[image]

And when I turn the pencil over:

[image]

To give a little more context, even without installing a driver, the eraser was working (without any driver installed before, the mac is new), it's like the driver didn't bring anything except the Wacom settings interface.

thenickdude, 2022-12-09:
With no driver installed the tablet works like a basic mouse (using relative positioning), but afterwards it should use absolute positioning (if you move the pen to the top left corner of the tablet, the cursor jumps to the top left of the screen, always).

Is this the original pen for the tablet? I've heard that incompatible pens can sometimes do this.

cladjidane, 2022-12-09:
The pen reacts as if the tablet was the screen: pen position = cursor on the screen, if I lift my hand and put it back down, for example, the cursor on the screen moves (jumps) to where my pen is (but only on the eraser side, the other side doesn't work). For the pencil, yes it is the pencil of the original tablet.

laptopleon, 2022-12-09:
I have two of these Intuos 2 tablets. A 'Platinum' one that I bought in 2004 that still works fine. Another deep blue one that I bought last year for a € 20 on a second hand site. Both with matching pen and mouse. The blue is older, looking at the serial numbers on the back of the tablet and mouse.

The pen of the blue tablet wouldn't work. Interestingly, the eraser side worked fine, just as with yours. 

I figured the previous owner had dropped it on the tip. This is the most commen way these pens get to malfunction. And then the whole set is basically useless.

I assumed a repair kit might be for sale online, or instructions how to fix it.

After quite a bit of searching online, I found a video, explaining that the tip is basically a very exactly tuned oscillator. When it falls on the tip, the tuning often simply gets off the mark, so to say. Hence, you can often fix this by re-calibrating it.

You can do this with a tiny screwdriver. Take the pen. Remove the 'button' on the side. Behind it are two holes where the button rests in and between them, in the middle, is a tiny screw.

I strongly suggest you first mark the position of the screw with a marker, so you know where it was. Then turn it half a turn and try if it works. Mine - sort of - worked again after 0,5 a clockwise turn with the screwdriver.

However, half the time, this pen / tablet combination gives quite a distorted effect. Movements look like as if the pen is on wavy paper. The pattern is very regular, but lines are no longer straight.

The weird thing is: When I move the pen 30 cm / 12 inch away and back, the movements are straight again! Until I move it back again.

The pen of the other, platinum tablet does not have this problem and works on both tablets. So there must be a difference.

Antyway, @cladjidane you might be able to fix your pen with this info. :)  Let us know.

cladjidane, 2022-12-12:
@laptopleon That's right! Thank you! I would never have found it, especially since there is no real info on this famous screw on the web. I had to play around a bit more to get a working pencil. I still think I have to improve because I find that there are some latency problems sometimes (not all the time).

Anyway it works fine, so nothing to do with the driver :)

Thanks again for the detailed message ^^

closed 2022-12-12 by cladjidane