thenickdude/wacom-driver-fix #105: CTH-670 still plagued by mapping panel issue in Ventura/M1 Max

joaofrgomes, 2022-12-10:
So yeah, it's basically what's in the description.

I haven't tested this latest pre-release on my 2012 MacBook Pro yet, where I even managed to get the mapping panel to open up after much frantic clicking even with the old version, but here on my Mac Studio I can't, for the life of me, get it to open.

I tried uninstalling and reinstalling the driver, also to no avail. Are there any files I might be missing, other than those the Wacom utility deletes on its own?

joaofrgomes, 2022-12-10:
By the way, I did manage to open the panel, but through the keyboard, because I have tabbing between all fields activated (it's still a leftover from my Windows user days over 19 years go).

So I guess I'm just **partially** affected by the issue.

joaofrgomes, 2022-12-10:
And yet another update: I can also reproduce the issue on Ventura running on my 2012 MacBook Pro, so this doesn't affect just M1-based Macs.

Wavewash, 2022-12-14:
> By the way, I did manage to open the panel, but only through the keyboard, because I have the Ctrl+F7 “tabbing between all fields” shortcut/setting activated (it's still a holdover from my Windows user days, i.e. a habit from over 19 years go).

Had the same issue on MacOS monterey with a 460-CTH. 

I wasn't able to open the mapping panel by mouse clicking but after tabbing to it I was able to get the panel to show. Appreciate this comment for pointing me to the right direction.

joaofrgomes, 2022-12-26:
Interesting! I tried the first set of instructions (the one which included clearing the search field) and it didn't work, but the second one did.

I'll still be using my method, at least until there's a fix, as it's the most obvious to me, muscle-memory-wise. Thank you for the heads-up, though, I'm sure it will be useful towards the development of said fix.

voskresla, 2023-03-27:
You can open `mapping` clicking on it by tablet pen for the first time.

jdnguyenswin, 2023-09-23:
> > By the way, I did manage to open the panel, but only through the keyboard, because I have the Ctrl+F7 “tabbing between all fields” shortcut/setting activated (it's still a holdover from my Windows user days, i.e. a habit from over 19 years go).
> 
> Had the same issue on MacOS monterey with a 460-CTH.
> 
> I wasn't able to open the mapping panel by mouse clicking. After clearing the search input I was able to get the mapping panel to show. Your comment about tabbing pointed me to what I think is the problem. The input is getting directed or captured by the search input perhaps? My hunch is this is related to focus being captured by the search input.
> 
> [image]
> 
> Steps to reproduce:
> 
> 1. Close out from system preferences -> wacom completely first.
> 2. Open system preferences -> wacom
> 3. Click pen tab
> 4. Mapping button should not display mapping panel
> 
> Steps to fix:
> 
> 1. Click the search input to populate it with the cursor
> 2. Press space and then tab
> 3. Click the circle x to clear the search input
> 4. Mapping button works!
> 
> Steps to reproduce another way that fixes it:
> 
> 1. Click the search input
> 2. Press space
> 3. Select another window/application that's not system preferences
> 4. Select system preferences which should still have the wacom pen tab open
> 5. Mapping button works!

This is truly amazing. Thank you so much, Wavewash.

Can anyone explain why the "search, space, tab, clear" works?

pohy, 2023-12-01:
> You can open `mapping` clicking on it by tablet pen for the first time.

Thanks, this has worked for me on Sonoma. 🙏

roboldham, 2023-12-04:
Hi folks... Sonoma here too... I canrt get any of the above methods to work.... the search dialgue is in a different place also.... anyone got any more tips for success?

[image]