thenickdude/wacom-driver-fix #106: Intuos 2 registers extra mouse clicks with Monterey 12.6.1 in the Mail App

jiceman, 2022-12-10:
First, thanks to Nick; I've been using your driver and sent a thank you donation when I found it. 

Filing as a bug, but really intended as a question.
I moved installed Monterey a couple of months ago. Since then, I notice that upon clicking once in Mail, I sometimes get the effect of two clicks or a double click. So, clicking on a mail message, sometimes pops it into a new window. Or, deleting one message sometimes deletes two. 

It only started recently. No change to Nick's wacom fixed driver. Only difference was going to MacOS 12.
I wondered if the mouse or tablet could be bad. But, I also use it on a PC and can't say I have seen a similar problem (yet). 
It is also odd, that I don't seem to see a problem except when using Mail.

Anyone else happen to be experiencing something similar?

laptopleon, 2022-12-10:
I'm not claiming to be an expert (at all), but I have been using Wacoms for three decades and even though Wacom is the brand with the best reputation, bugs are quite common. Also all kinds of problems tend to arise, that are or are not caused by Wacom's drivers, other software / os'es or a combination of both.

The problem you claim sounds a lot like the problem in this video: https://www.youtube.com/watch?v=LlrWWDFjT7E

Have you tried uninstalling the Wacom drivers / software / preference files, including the files that are left when you uninstall via the Wacom software button? (Also, see the video + video description).

jiceman, 2022-12-10:
Looks somewhat similar. But a different device. Still, could be related. I have not seen the problem for years though, only noticed past few months about the time I upgraded OS.

I don't have all the settings available that he changes, but I can experiment a bit. I tried chanbging the double click speed and it was not a solution for me.

thenickdude, 2022-12-11:
Wacom "drivers" are regular user-level apps, so any kernel panics could only come from macOS bugs, not Wacom bugs.

jiceman, 2022-12-17:
I've observed the double click using other programs besides Mail. And, also a few times on a Windows computer. So, perhaps it's a hardware issue. I've ordered a replacement (used) mouse and will see if that fixes the problem.

jiceman, 2022-12-30:
Found a replacement mouse. So far, no double clicks. I'll test for a couple of more days and then close this. Looks like it was due to faulty hardware.

closed 2023-01-02 by jiceman

Sleepful, 2026-09-27:
Seems like pth-651 has been rendered obsolete by Wacom in Mac Tahoe, latest working driver was 6.4.9-2, not the latest published driver which is 6.4.10... It seems blatantly intentional at this point