thenickdude/wacom-driver-fix #110: Adjustment with Intuos4 - PTK-640 possible?

bunducafe, 2023-02-24:
I tried some workarounds but with macOS ventura the PTK-640 has now stopped working completely. Is there a solution for that tablet, too? That would be highly appreciated.

bunducafe, 2023-02-25:
Okay, found a workaround for my problem with the Intuos4. To be found [here at the Wacom official site](https://support.wacom.com/hc/en-us/articles/1500006264541-How-do-I-uninstall-manually-and-re-install-the-Wacom-driver-on-macOS-for-a-Pen-Tablet-Pen-Display-or-Pen-Computer-)

After deleting everything manually and reinstalled the old driver also the Intuos4 seems to work with macOS Ventura.

closed 2023-02-25 by bunducafe

jeffreylm, 2023-02-25:
When you say old. Which old version exactly? And are you saying only install the old drivers from Wacom ,and ignore this wacom driver fix here on Git-Hub?

bunducafe, 2023-02-25:
"Old" means the "newest" recommended driver. It is also stated in the link and for the Intuos4 it was the driver 6.3.41-2

The fix here was only for Intuos1-3 and is also stated in the readme. Luckily I stumbled upon the mentioned workaround because otherwise I was already buying a new device which is now not necessary.

jeffreylm, 2023-02-25:
OOh, ok thank you for responding. I literally have a perfectly good Intuos PTK-640, that works fine under Mojave, and stopped working since catalina, and have it in a box. And I tried it a few days ago under Montery, and i din't work. Was so fustrated and tought, there must be someone on the planet that has a solution. As I have never had a problem with computers that wasn't solved by someone on the internet. And I mean this has been so since the 90's. I meant never had a unsolved problem. Sometimes it takes weeks, months or years, but always gets solved.

sevendemagnus, 2023-08-23:
Hello sir @jeffreylm have you gotten your Intuos 4 to work with macOS Ventura? If yes, if you could send a link or instructions on how you did it, that'd be a great. I have an Intuos 4 Wireless version.

God bless, thank you in advance.