thenickdude/wacom-driver-fix #113: Any support being added for newly EOL products? (g. Intuos 5)

B00Berry, 2023-04-02:
Self explanatory.  Wacom requires drivers 6.4+ for Ventura compatibility, and both of my Intuos 5 tablets are stuck at 6.3.46-2 as they have been added to the dreaded [End-of-Life products list](https://support.wacom.com/hc/en-us/articles/4415799565719-What-products-can-no-longer-be-serviced-by-Wacom-).  Which is BS as you all know, because these products have plenty of life left in them if it weren't for forced software obsolescence.

thenickdude, 2023-04-02:
That's the official statement on compatibility, but I understand their older drivers work fine for Intuos 5 on Ventura.

If you accidentally install the newest driver you have to manually uninstall it using Wacom Utility to get rid of it first.

B00Berry, 2023-04-03:
> That's the official statement on compatibility, but I understand their older drivers work fine for Intuos 5 on Ventura.
> If you accidentally install the newest driver you have to manually uninstall it using Wacom Utility to get rid of it first.

I haven't changed my driver.  6.3.46-2 was working just fine for me last week on Mojave.  Now that I have updated to Ventura, I get very frequent dead input with eh pen tip.  My current workaround is to touch the eraser end of the pen to the tablet, which never seems to die.  Sometimes it takes a few touches of the eraser to get the pen to work again... or I could just be fooling myself that it has any effect at all, and I'm just giving myself something to do while the pen would become responsive again on it's own.

Regardless, this issue began after the Ventura update, and occurs sometimes quite frequently (more than once per minute).

B00Berry, 2023-04-03:
> That's the official statement on compatibility, but I understand their older drivers work fine for Intuos 5 on Ventura.
> If you accidentally install the newest driver you have to manually uninstall it using Wacom Utility to get rid of it first.

Thanks, I appreciate that info.  However, I haven't changed my driver.  6.3.46-2 was working just fine for me last week on Mojave.  Now that I have updated to Ventura, I get very frequent dead input with the pen tip.  My current workaround is to touch the eraser end of the pen to the tablet (which always seems responsive).  Sometimes it takes a few touches of the eraser to get the pen tip to work again... or I may be fooling myself that it has any effect at all, and I'm just giving myself something to do while the pen would become responsive again on it's own.

Regardless, this issue began after the Ventura update, and occurs sometimes quite frequently (more than once per minute).

B00Berry, 2023-04-07:
Upon more use with Ventura, it's apparent that the Intuos 5 pen-tip maintains input as long it's not lifted from the tablet surface.  When lifted and then returned to the tablet, the tip becomes a dead input for a seemingly random amount of time (although the eraser end still works reliably).  Very bizarre, and frustrating.  I've found myself using the eraser end as a pen tip in frustration, & relying on the keyboard for modifiers instead of the pen buttons....  

I really don't want to replace an expensive large tablet with no physical issues. It's killing my productivity though.