thenickdude/wacom-driver-fix #118: Apple Silicon, Rosetta & reverse engineered/clean room drivers

joaofrgomes, 2023-06-30:
Hi!

I've been happily using your fixed drivers for years but, as someone who's been through an ISA transition on the Mac before, I know fully well Apple wants to deprecate Rosetta 2 ASAP.

When (not if) they do so, even your fixed drivers will cease to work altogether with all Apple Silicon-based systems, and it will not be feasible to just stay on older versions of macOS for long; as such, is there any chance you or some other elite developer might be able to reverse-engineer Wacom's drivers and come up with new, Apple Silicon/Universal ones that might keep our tablets working until their natural deaths?

All the best and, once again, kudos for your great work!

thenickdude, 2023-06-30:
There's a project OpenTabletDriver which does this, but they don't have pen pressure support on macOS yet.

Bottacco, 2024-01-14:
While the OpenTablet is still only available in binary form for macOS x64 (you may try to compile it natively on Mx machines, but I have not been able to do it, too many errors that I cannot understand how to solve) it is a user mode driver, so it may work via Rosetta 2.
Lets hope that native ARM support comes soon.

joaofrgomes, 2025-09-14:
Quick update: After some chatting with the OTD devs at their Discord server, it seems that their drivers will run headless, i.e. without a GUI, on Apple Silicon, and have to be built/packaged specifically for that platform (osx-arm64) with the .NET 8.0 SDK. It's an involved, not at all user-friendly process, but it should get the job done.

They are also dependent on the Eto framework supporting Apple Silicon, and will not look into making native Apple Silicon builds until their next major version (IIRC, that will be 0.7), but already have the underpinnings for cross-platform builds in place.

Yes, they are way more optimistic than me about Apple keeping Rosetta 2 working, which may explain why they aren't prioritizing this as much as I would expect, but all signs point to the transition to be feasible and timely if Apple does go the most aggressive route. At worst, it may keep some users from updating to macOS 27 straight away and, if it's still the only option available, I'll be sure to test their latest x86-64 build of OTD (and, of course, nickthedude's fixed driver, which will likely be my first go-to option until they stop being viable) next year with the first Public Beta of that upcoming version and sound the alarm if it doesn't work at all.