thenickdude/wacom-driver-fix #122: Intuos 1 on Ventura

fvlasie, 2023-08-12:
Hello everyone!

Has anyone been able to get the Intuos 1 driver working in Ventura? It basically works for me (amazing, thanks Nick!) except that I cannot get the pen to go into mouse mode. I have two side-by-side ultrawide ratio monitors and this makes the horizontal movement of the pen stretch significantly compared to the vertical...so I end up drawing a squashed oval when I want a circle. I don't see any screen mapping options either. 

Has my 23 year old tablet finally reached the limits of life? :(

schubidubi, 2023-08-19:
I haven't tested it very thoroughly yet because I've rarely used it in the past 10 years anyway, but yeah, my 23 years old Intuos 1 still essentially works on brand new MBA 15". Amazing, thanks to Nick's hack! :)
And so does the CTE-430, although I can't get the popup menu working.

Bottacco, 2024-01-14:
> Hello everyone!
>
> Has anyone been able to get the Intuos 1 driver working in Ventura? It basically works for me (amazing, thanks Nick!) except that I cannot get the pen to go into mouse mode. I have [...]

The drivers are only X64, so I don't know if they are going to work on M1, M2, M3 or any ARM variant that Apple comes out with because if you look inside the installers there are several kexts and M machines won't load kexts that are not compiled for their architecture.