thenickdude/wacom-driver-fix #123: M1 Mac - Intuos 4 or Bamboo devices not found

pauljvallance, 2023-08-21:
More a question really. I cannot get my M1 Mac Mini (OS 12.6.8) to recognise either of my tablets using the software you suggest.
I have an Intuos 4 (PTK-440) and a Bamboo (CTL-490).

Any help or workarounds would be much appreciated.

Paul

Bottacco, 2024-01-14:
If you look the main page it clearly states what models are supported and it only supports Intuos 1,2 or 3. For Intuos 4 you need to use the official driver from Wacom's website. The same for the Bamboo CTL-490.

schroef, 2025-06-04:
> If you look the main page it clearly states what models are supported and it only supports Intuos 1,2 or 3. For Intuos 4 you need to use the official driver from Wacom's website. The same for the Bamboo CTL-490.

He commented himself on another post, stating incidentally it seems to work. But weird he doesn't state it in the support though 

https://github.com/thenickdude/wacom-driver-fix/issues/57#issuecomment-745094383