thenickdude/wacom-driver-fix #126: Bamboo CLT-470 on macOS 14 (Sonoma)

dgobnto, 2023-11-08:
Hi,

I was forced to upgrade to macOS 14 (Sonoma) a few days ago, as my Macbook Pro 14'' M1 Pro died and had the logic board replaced.

Since then, I get the Preferences panel warning "Waiting for synchronization" and then "There is a problem with your tablet driver. Please reboot your system […]".

I only have your v5.3.7-6 installed and see the PenTableDriver.app in Input Monitoring. Already tried uninstalling preferences, installing driver again but nothing seems to work.

Thank you for any support possible.

SgtSalmon, 2023-11-21:
Works on my sonoma install (with a few bugs), sounds like you have the unmodified preference panel installed. Go to /Library/PreferencePanes/ to find the wacom preference pane, delete this, than run the package installer again.

dgobnto, 2023-11-21:
It seems that my issue was related to 'Nicholas Sherlock' item being disabled in the Login Items page. What does it do?

closed 2023-11-21 by dgobnto

SgtSalmon, 2023-11-21:
It launches the driver dameon, usually this would be called wacom, but it goes off of the signature name, which is Nick, the developer.