thenickdude/wacom-driver-fix #131: Does someone know if this driver will work on the Ventura or Sonoma?

nyriande, 2024-02-01:
I have the CTH-670 and would like to update my Mac to Ventura or Sonoma but has someone tried if the driver works for that?

liam-k, 2024-02-18:
I just tried it and it works on Sonoma for my CTH-470! The only thing that doesn’t work is assigning a keystroke or a modifier to one of the buttons on the tablet. But for now I’m just glad my Bamboo isn’t a useless brick anymore, so... thank you @thenickdude!

SpyderBrian, 2024-05-06:
> I just tried it and it works on Sonoma for my CTH-470! The only thing that doesn’t work is assigning a keystroke or a modifier to one of the buttons on the tablet. But for now I’m just glad my Bamboo isn’t a useless brick anymore, so... thank you @thenickdude!
> 
> EDIT: I have to correct myself unfortunately. Many things don't work in terms of customization. Basically any option in the preference pane that has a pop-up or has something to do with with custom shortcuts, custom clicks, custom actions is not available under Sonoma. Still, it's much better than nothing.

Same for me on Sonoma with a CTH-460 on M3 iMac.

cajhin, 2024-07-12:
Graphire 1 works on M2 MBA with MacOS 14.5
Eraser doesn't work in Affinity, otherwise it's just fine. 25 years old hardware, on the lastest MBA, I love it :)