thenickdude/wacom-driver-fix #135: Settings menu missing Intuos 1 on Sonoma 14.3.1 (23D60)

CainArg, 2024-05-15:
Hi!

first of all, THANKS!!!!
I'm using Install-Wacom-Tablet-6.1.6-4-patched

Under Tool / Intuos Pen / Pen:
Can't get KEYSTROKE dialog on Pen's buttons so you can't define keys combinations

[image]

Under Tool / Intuos Pen / Eraser:
Can't change the ERASE drop menu...

[image]

Thanks!!!!!

vlad15374, 2024-06-15:
Also have same problems with Wacom 3.
Please somebody give some clue how we can fix or workaround this?

vlad15374, 2024-06-15:
Workaround:
I simply make a settings-backup via "wacom Tablet Utility.app" and then Restore from that Backup. After this driver is relaunching and ask to give him permissions, like on first installation. After this — woala! **all settings and popups for editing Keystrokes and so on is working!**