thenickdude/wacom-driver-fix #140: Intuos 3 - SO CLOSE, but click not working

Hepworks, 2024-11-21:
Oh man but this feels so close.

I now have the full range of motion, mapping, etc.

The driver can sense the click pressure, but I can't actually click on anything

any help greatly appreciated!

[https://drive.google.com/file/d/1DM-LYxKGsTNCKSaG8_X1tphBt5DiFTtO/view?usp=sharing](url)

MattrCoUk, 2025-03-30:
Works fine for me on Ventura 13.7.4 ! 
Unfortunately driver crashes when I open Adobe Illustrator, but key features of tablet work fine with Figma (not sure about pressure tho). Many Thanks @thenickdude !

schroef, 2025-06-04:
I'm having same isseu in Sequoia. I got a good mapping now, it registers pen pressure, but no click and no buttons respond

Ameranthi, 2025-07-04:
Hi @schroef @Hepworks I had this problem previously. You can resolve this issue by allowing WacomTabletSpringBoard access to the accessibility settings via this path: Standard Mac Settings > Privacy & Security > Accessibility click toggle on

On Mac the Wacom stylus uses the accessibility features to enable the clicking function in the Operating System

[image]

@thenickdude  please resolve issue

schroef, 2025-07-04:
@Ameranthi 

I removed everything related to this repo and tried some older releases. They work. Only thing I now have, it takes 1-2 minutes after a restart to get it to work. Plus, I get a warning popup which I need okay forz kinda weird because everything is checked "allow"

Also after sleepy it looses connection, I need to unplug the device and plug it back it. But all of this, is using original Wacom drivers now